KumarRobotics/imu_vn_100 #3: Timeout

versatran01, 2015-11-04:
need to fix

versatran01, 2015-11-05:
@kartikmohta what is your temporary solution?

versatran01, 2016-02-21:
The timeout thing is still happening.

kartikmohta, 2016-02-21:
Yes it does.

versatran01, 2016-02-21:
:dizzy_face:

andre-nguyen, 2016-02-23:
How do you reproduce this?

versatran01, 2016-02-23:
It's hard to reproduce with the driver itself. But when we run our cam_imu_sync, it reports timeout error from the sdk, it happens pretty often but I'm not able to locate the issue.

andre-nguyen, 2016-02-23:
If it's an SDK error code, I've noticed that the manual error codes don't match with the SDK error codes. Page 32 of the manual:

[image]

whereas the SDK says

[image]

I'm wondering which one is correct. When I ran into the permission error because I tried to start the driver on an unconnected serial port (of the VN100) I felt the `invalid parameter` error made more sense than a `permission denied` error which is supposed to be triggered when you try to write a read only register.

andre-nguyen, 2016-02-24:
I emailed VectorNav about this issue and their support guy sent me a link to a new version of the library:

> Hello Andre,
> 
> Thank you for your email. We are in the process of releasing some major updates to the programming library. Below is a link to a beta version of the new library. Please let me know if you continue to encounter issues and I will put you in touch with our developers.
> 
> Windows - http://www.vectornav.com/Downloads/proglib/vnproglib-1.1.0.115.exe
> Linux - http://www.vectornav.com/Downloads/proglib/vnproglib-1.1.0.115.tar.gz

The architecture looks significantly different from the old version but it confirms my suspicion that the error codes in the 0.3.2 library are wrong:

See the types.h files in `vnproglib-1.1.0.115\cpp\include\vn\protocol\uart`

```C++
/// \brief Errors that the VectorNav sensor can report.
enum SensorError
{
	ERR_HARD_FAULT = 1,					///< Hard fault.
	ERR_SERIAL_BUFFER_OVERFLOW = 2,		///< Serial buffer overflow.
	ERR_INVALID_CHECKSUM = 3,			///< Invalid checksum.
	ERR_INVALID_COMMAND = 4,			///< Invalid command.
	ERR_NOT_ENOUGH_PARAMETERS = 5,		///< Not enough parameters.
	ERR_TOO_MANY_PARAMETERS = 6,		///< Too many parameters.
	ERR_INVALID_PARAMETER = 7,			///< Invalid parameter.
	ERR_INVALID_REGISTER = 8,			///< Invalid register.
	ERR_UNAUTHORIZED_ACCESS = 9,		///< Unauthorized access.
	ERR_WATCHDOG_RESET = 10,			///< Watchdog reset.
	ERR_OUTPUT_BUFFER_OVERFLOW = 11,	///< Output buffer overflow.
	ERR_INSUFFICIENT_BAUD_RATE = 12,	///< Insufficient baud rate.
	ERR_ERROR_BUFFER_OVERFLOW = 255		///< Error buffer overflow.
};
```

versatran01, 2016-02-24:
I will probably leave the current one as it is, and wait for their release of the new library and switch to that.

andre-nguyen, 2016-02-24:
Agreed, I don't see the benefit of changing this. The question then becomes, why are you guys getting a checksum error when synchronizing with the cameras...

versatran01, 2016-02-24:
Oh, I must have described the problem incorrectly. The timeout only happens at when the imu initialization step. When the driver tried to talk to the imu, after that there's no error whatsoever.

kartikmohta, 2016-03-08:
Doing a reset at the end when disconnecting solves this issue since it resets the baud rate in the imu to the default.
Haven't seen this happening on the system after fixing the runtime error/exception in KumarRobotics/cam_imu_sync@fafc2b7218cd40ee2a5a8b8360ce0815dc26074a. I guess the disconnect method wasn't properly called when it exited with the runtime error.

closed 2016-03-08 by versatran01

Azimath, 2016-07-17:
I've been having a few issues with timeouts trying to do the initial connection where sometimes my vn100 would start at 9600 baud instead of 115200. Implementing a quick and dirty loop around line 143 of imu_vn_100.cpp that tries connecting at various supported baud rates seems to have made my system more reliable.

`  ROS_DEBUG("Connecting to device");
  unsigned int old_baudrate;
  unsigned int bauds[3] = {9600, 115200, 921600};
  for(int i = 0; i < 3; i++)
  {
    if(vn100_connect(&imu_, port_.c_str(), bauds[i]) == VNERR_NO_ERROR)
    {
        old_baudrate = bauds[i];
        break;
    }
  }
  ros::Duration(0.5).sleep();
  ROS_INFO("Connected to device at %s", port_.c_str());
  
  ROS_INFO("Default serial baudrate: %u", old_baudrate);`