KumarRobotics/imu_vn_100 #7: /imu: VN: Permission denied

MahsaP, 2016-05-27:
Hello,

I downloaded the package. When I run:
`roslaunch imu_vn_100 vn_100_cont.launch
`
I get the following result in the terminal. "/imu: VN: Permission denied". Would you please help me to solve the problem.

```
process[rosout-1]: started with pid [13597]
started core service [/rosout]
process[imu-2]: started with pid [13607]
[ INFO] [1464357306.167245353]: Sync out rate: 20
[ INFO] [1464357306.748363389]: Connected to device at /dev/ttyUSB0
[ INFO] [1464357306.765977143]: Default serial baudrate: 115200
[ INFO] [1464357306.766011559]: Set serial baudrate to 921600
[ INFO] [1464357307.776820987]: Connected to device at /dev/ttyUSB0
[ INFO] [1464357307.783369148]: New serial baudrate: 921600
[ INFO] [1464357307.788441741]: Fetching device info.
[ INFO] [1464357307.793557811]: Model number: VN-100T-CR
[ INFO] [1464357307.798613813]: Hardware revision: 7
[ INFO] [1464357307.803668418]: Serial number: 0100017300
[ INFO] [1464357307.808746557]: Firmware version: 2.0.0.0
[ INFO] [1464357307.808777854]: Set Synchronization Control Register (id:32).
[ INFO] [1464357307.837879233]: /imu: VN: Permission denied
[imu-2] process has finished cleanly
```
Thank you.
Mahsa

Azimath, 2016-07-17:
On line 217 in src/imu_vn_100.cpp, try changing `BINARY_ASYNC_MODE_SERIAL_2` to `BINARY_ASYNC_MODE_SERIAL_1` and see if that changes anything. I had the same error and this fixed it for me.

mplefort, 2016-08-01:
had the same error. Fixed it for me too. Thanks!!

MahsaP, 2016-08-23:
@Azimath Thanks a lot. It is working now.

closed 2016-08-23 by MahsaP

ChanghyeonKim93, 2017-07-09:
Azimath's answer is awesome!!!! It works well!!!!

krish007k, 2019-04-30:
it is not working in my system. can u tell me any other possible solution for that problem
.