KumarRobotics/imu_vn_100 #13: Error using this driver!!!

vik748, 2017-10-05:
I am trying to use this drive on a VN-100 in ROS Kinetic.  I am able to compile everything correctly, however I get an error when I run it.  The VN-100 is default out of the box, so I am setting the baudrate to 115200 in the config file.  However the IMU rate of 100hz, throws a permission error.  So I set the IMU rate to 20 and here is what I see...
`[ INFO] [1507221665.943186795]: Sync out rate: 100
[ INFO] [1507221666.559237034]: Connected to device at /dev/ttyUSB0
[ INFO] [1507221666.591308742]: Default serial baudrate: 115200
[ INFO] [1507221666.591380828]: Set serial baudrate to 115200
[ INFO] [1507221667.623906113]: Connected to device at /dev/ttyUSB0
[ INFO] [1507221667.651121672]: New serial baudrate: 115200
[ INFO] [1507221667.671867168]: Fetching device info.
[ INFO] [1507221667.687437961]: Model number: VN-100S-CR
[ INFO] [1507221667.702816139]: Hardware revision: 7
[ INFO] [1507221667.718275868]: Serial number: 0100027862
[ INFO] [1507221667.733840069]: Firmware version: 2.0.1.0
[ INFO] [1507221667.733908211]: Set Synchronization Control Register (id:32).
[ INFO] [1507221667.800836884]: Setting IMU rate to 20
`
When I run `rostopic list` I am able to see the topics, however `rostopic echo /imu/imu` does't show me anything.

Any ideas where I am going wrong?

Thanks,
Vikrant

kartikmohta, 2017-10-05:
After connecting the IMU to the computer, can you try running the following bash script before launching the driver and then check if it works.
```bash
# Reduce latency in the FTDI serial-USB kernel driver to 1ms
# This is required due to https://github.com/torvalds/linux/commit/c6dce262
for file in $(ls /sys/bus/usb-serial/devices/); do
  value=`cat /sys/bus/usb-serial/devices/$file/latency_timer`
  if [ $value -gt 1 ]; then
    echo "Setting low_latency mode for $file"
    sudo sh -c "echo 1 > /sys/bus/usb-serial/devices/$file/latency_timer"
  fi
done
```

vik748, 2017-10-06:
Nope, this didn't help.  Could it be because the driver disconnects and reconnects??

> [DEBUG] [1507297569.144360156]: remap: __name => imu
> [DEBUG] [1507297569.144404706]: remap: __log => /home/vik748/.ros/log/1a01016a-a9f0-11e7-ac0c-c4b301b9ee07/imu-1.log
> [DEBUG] [1507297569.145478927]: Adding tcp socket [6] to pollset
> [DEBUG] [1507297569.145548390]: UDPROS server listening on port [37692]
> [DEBUG] [1507297569.148567181]: Started node [/imu], pid [26213], bound on [vik-MacBookPro], xmlrpc port [33191], tcpros port [46633], using [real] time
> [DEBUG] [1507297569.149986413]: Subscribed to parameter [/imu/diagnostic_period]
> [DEBUG] [1507297569.150507241]: XML-RPC call [getParam] returned an error (-1): [Parameter [/imu/diagnostic_period] is not set]
> [DEBUG] [1507297569.150544912]: Caching parameter [/imu/diagnostic_period] with value type [0]
> [ INFO] [1507297569.161645786]: Sync out rate: 20
> [DEBUG] [1507297569.161677070]: Connecting to device
> [DEBUG] [1507297569.405763161]: Accepted connection on socket [6], new socket [11]
> [DEBUG] [1507297569.406222623]: Adding tcp socket [11] to pollset
> [DEBUG] [1507297569.406306988]: TCPROS received a connection from [127.0.0.1:48972]
> [DEBUG] [1507297569.406625066]: Connection: Creating TransportSubscriberLink for topic [/rosout] connected to [callerid=[/rosout] address=[TCPROS connection on port 46633 to [127.0.0.1:48972 on socket 11]]]
> [ INFO] [1507297569.777053528]: Connected to device at /dev/ttyUSB0
> [ INFO] [1507297569.813030507]: Default serial baudrate: 115200
> [ INFO] [1507297569.813093177]: Set serial baudrate to 115200
> [DEBUG] [1507297569.838973373]: Disconnecting the device
> [DEBUG] [1507297570.344511685]: Reconnecting to device
> [ INFO] [1507297570.846133019]: Connected to device at /dev/ttyUSB0
> [ INFO] [1507297570.871749153]: New serial baudrate: 115200
> [ INFO] [1507297570.897495788]: Fetching device info.
> [ INFO] [1507297570.913054307]: Model number: VN-100S-CR
> [ INFO] [1507297570.929120745]: Hardware revision: 7
> [ INFO] [1507297570.944206231]: Serial number: 0100027862
> [ INFO] [1507297570.959554188]: Firmware version: 2.0.1.0
> [ INFO] [1507297570.959605623]: Set Synchronization Control Register (id:32).
> [ INFO] [1507297571.026769715]: Setting IMU rate to 50
> [DEBUG] [1507297574.228761982]: Accepted connection on socket [6], new socket [10]
> [DEBUG] [1507297574.229023491]: Adding tcp socket [10] to pollset
> [DEBUG] [1507297574.229080174]: TCPROS received a connection from [127.0.0.1:49006]
> [DEBUG] [1507297574.229174922]: Connection: Creating TransportSubscriberLink for topic [/imu/imu] connected to [callerid=[/rostopic_26253_1507297574031] address=[TCPROS connection on port 46633 to [127.0.0.1:49006 on socket 10]]]

kartikmohta, 2017-10-06:
Can you try the change in #11, that seems like the a similar problem.

vik748, 2017-10-06:
As suggested, this problem is similar to #11 and was fixed as suggested in there.
@kartikmohta thanks for your help

closed 2017-10-06 by vik748

tranz3r, 2019-01-20:
Hi, I am using the sensor and it has permission problems denied, I have tried the proposed solutions changing the parameters of BINARY_ASYNC_MODE to port 1, however I still have no answer. Please, if you can help me, I'm a little desperate for not coming up with the solution. I'm work with ROS melodic and Ubuntu 18.04.1 LTS. On the terminal display this:
tranz3r@ubuntu:~/hexapod_ws/imu_vn_100/launch$ roslaunch vn_100_cont.launch
... logging to /home/tranz3r/.ros/log/d3cd6cd2-1c7c-11e9-a1a7-000c29f40dd6/roslaunch-ubuntu-6837.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43541/

SUMMARY
========

PARAMETERS
 * /imu/baudrate: 115200
 * /imu/binary_async_mode: 1
 * /imu/binary_output: True
 * /imu/enable_mag: True
 * /imu/enable_pres: True
 * /imu/enable_rpy: False
 * /imu/enable_temp: True
 * /imu/frame_id: imu
 * /imu/imu_compensated: False
 * /imu/imu_rate: 20
 * /imu/port: /dev/ttyUSB0
 * /imu/sync_pulse_width_us: 1000
 * /imu/sync_rate: 20
 * /imu/vpe/enable: True
 * /imu/vpe/filtering_mode: 1
 * /imu/vpe/heading_mode: 1
 * /imu/vpe/tuning_mode: 1
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    imu (imu_vn_100/imu_vn_100_node)

ROS_MASTER_URI=http://localhost:11311

process[imu-1]: started with pid [6852]
[ INFO] [1547965937.600805061]: Sync out rate: 20
[ INFO] [1547965937.602044458]: /imu: VN: Permission denied
[imu-1] process has finished cleanly
log file: /home/tranz3r/.ros/log/d3cd6cd2-1c7c-11e9-a1a7-000c29f40dd6/imu-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

ke-sun, 2019-01-20:
Have you checked if you have the correct permission to the corresponding file under /dev?

tranz3r, 2019-01-20:
> Have you checked if you have the correct permission to the corresponding file under /dev?

Hello, thanks for the answer. I am very new to ROS. I have verified the permissions by the command 
ls -l inside the imu_vn_100 directory, which I throw the following:
tranz3r @ ubuntu: ~ / hexapod_ws / imu_vn_100 $ ls -l
total 44
-rw-r - r-- 1 tranz3r tranz3r 1689 Jan 15 18:58 CMakeLists.txt
drwxr-xr-x 3 tranz3r tranz3r 4096 Jan 19 23:33 include
drwxr-xr-x 2 tranz3r tranz3r 4096 Jan 20 00:32 launch
-rw-r - r-- 1 tranz3r tranz3r 11358 Jan 15 18:58 LICENSE.txt
-rw-r - r-- 1 tranz3r tranz3r 624 Jan 15 18:58 package.xml
-rw-r - r-- 1 tranz3r tranz3r 6179 Jan 15 18:58 README.md
drwxr-xr-x 2 tranz3r tranz3r 4096 Jan 19 23:33 src
drwxr-xr-x 4 tranz3r tranz3r 4096 Jan 19 23:33 vncpplib

  Afterwards, change the permissions of the file vn_100_cont.launch by means of the command: chmod 777 vn_100_cont.launch
After that, I went back to execute roslaunch vn_100_cont.launch, with the same results.
Could you please guide me to know what I could be doing wrong? Thanks a lot.

shahvi, 2019-01-21:
@tranz3r you need to make sure that the device '/dev/ttyUSB0' has the correct permissions for your user account. The following will ensure it will work everytime:

1. As Sudo create a file under /etc/udev/rules.d called 50-VN-100.rules with the following
`KERNEL=="ttyUSB[0-9]*", ACTION=="add", ATTRS{idVendor}=="1d6b",ATTRS{idProduct}=="0002", MODE="0666", GROUP="dialout"
`
Also, Ubuntu's default latency of 16ms seems to cause issues at higher rates, that can be solved by adding the following UDEV rule as 49-USB-LATENCY.rules
`ACTION=="add", SUBSYSTEM=="usb-serial", DRIVER=="ftdi_sio", ATTR{latency_timer}="1"
`
2. Once the rules have been added, to get udev to recognize the rule, run the following command:
`sudo udevadm control --reload-rules && sudo service udev restart && sudo udevadm trigger`
3. Finally unplug and replug the VN-100 to have it work with the new rule.

tranz3r, 2019-01-22:
> @tranz3r you need to make sure that the device '/dev/ttyUSB0' has the correct permissions for your user account. The following will ensure it will work everytime:
>
>     1. As Sudo create a file under [...]

> @tranz3r you need to make sure that the device '/dev/ttyUSB0' has the correct permissions for your user account. The following will ensure it will work everytime:
>
>     1. As Sudo create a file under [...]

Hi all!
I have implemented the proposed solutions and finally I have been able to communicate my VN-100 correctly with ROS, thank you very much for the help and availability.