KumarRobotics/imu_vn_100 #14: /imu: VN: Permission denied

shubham294, 2017-10-08:
Hi,

I tried running the node through the given roslaunch but strangely it is throwing permission errors even when it could initially talk to the sensor. It's different from the issue #7 as quite a lot has changed since then. This problem raised when I pulled in the latest master today. I was using the earlier version without the VPE controls.
Logs are attached below.

`[ INFO] [1507493246.354727802]: Sync out rate: 20
[ INFO] [1507493246.855516924]: Connected to device at /dev/ttyUSB0
[ INFO] [1507493246.894644059]: Default serial baudrate: 115200
[ INFO] [1507493246.894686756]: Set serial baudrate to 921600
[ INFO] [1507493247.931730653]: Connected to device at /dev/ttyUSB0
[ INFO] [1507493247.949351739]: New serial baudrate: 921600
[ INFO] [1507493247.969925247]: Fetching device info.
[ INFO] [1507493247.985199319]: Model number: VN-100S-CR
[ INFO] [1507493248.000538898]: Hardware revision: 7
[ INFO] [1507493248.015881901]: Serial number: 0100022358
[ INFO] [1507493248.031221457]: Firmware version: 2.0.1.0
[ INFO] [1507493248.031265114]: Set Synchronization Control Register (id:32).
[ INFO] [1507493248.066934842]: Default VPE enable: 1
[ INFO] [1507493248.066981043]: Default VPE heading mode: 1
[ INFO] [1507493248.067001717]: Default VPE filtering mode: 1
[ INFO] [1507493248.067019919]: Default VPE tuning mode: 1
[ INFO] [1507493248.067053520]: Setting VPE MagnetometerBasicTuning BaseTuning (4.000000, 4.000000, 4.000000)
[ INFO] [1507493248.067078179]: Setting VPE MagnetometerBasicTuning AdaptiveTuning (5.000000, 5.000000, 5.000000)
[ INFO] [1507493248.067102500]: Setting VPE MagnetometerBasicTuning AdaptiveFiltering (5.500000, 5.500000, 5.500000)
[ INFO] [1507493248.082273786]: Setting VPE AccelerometerBasicTuning BaseTuning (5.000000, 5.000000, 5.000000)
[ INFO] [1507493248.082320937]: Setting VPE AccelerometerBasicTuning AdaptiveTuning (3.000000, 3.000000, 3.000000)
[ INFO] [1507493248.082342291]: Setting VPE AccelerometerBasicTuning AdaptiveFiltering (4.000000, 4.000000, 4.000000)
[ INFO] [1507493248.128296534]: Streaming #1: BG1_QTN|BG1_SYNC_IN_CNT|BG1_IMU
[ INFO] [1507493248.128346455]: Streaming #3: BG3_UNCOMP_MAG|BG3_TEMP|BG3_PRES
[ INFO] [1507493248.153080834]: /imu: VN: Permission denied
[imu-2] process has finished cleanly
`

kartikmohta, 2017-10-08:
Can you try again after changing the `binary_async_mode` to 1 on [Line 17](https://github.com/KumarRobotics/imu_vn_100/blob/1f05426d29580eb9fecc1fb68ed0f89280f7548f/launch/vn_100_cont.launch#L17) of the launch file?

shubham294, 2017-10-09:
Thanks! It worked! :)

closed 2017-10-09 by shubham294