KumarRobotics/imu_vn_100 #15: can not obtain high freuqnecy

snakehaihai, 2017-10-10:
At first when i use the driver, if i set hz rate to be 400. it will be close
but after using like 1 month. if i set 400 now, the rate is only 100 to 300 only.
why this could happen?

kartikmohta, 2017-10-10:
I've added a possible reason for this in the FAQ: https://github.com/KumarRobotics/imu_vn_100#faq
Can you try it and see if it works for you?

snakehaihai, 2017-10-11:
Yes thats the issue. b4 if target is 400, it was like 150- 250. now if 400, its 380 to 390.  is there any possible way to increase it to 400? or make it less oscillate

kartikmohta, 2017-10-11:
We've never gotten it to work at full 400Hz either, we get around 380-390Hz maximum. For our purpose, we run it at 200Hz and it works perfectly.

snakehaihai, 2017-10-11:
But i 'm really interested to know why my pc initially can get exact 400 hz. Thanks for the help. now i understand why my visual inertial navigation might drift. So it is all because of the kernel delay for FDTI cable

kartikmohta, 2017-10-11:
Don't know how you got it to work at 400Hz before, so can't help you there.
For the VIO algorithm I think that 200Hz (or maybe even 100Hz) IMU data would be sufficient, you don't really need to run the IMU at 400Hz.

snakehaihai, 2017-10-13:
Ok it is ture. 400 hz is only for those fast moving UAVs. for small MAV 200 hz is fine. Thanks

closed 2017-10-13 by snakehaihai