KumarRobotics/imu_vn_100 #16: Noisy Linear Acceleration

shubham294, 2017-10-31:
Hi,

I enabled streaming of gravity-removed linear acceleration by adding BG5_LINEAR_ACCEL_BODY to Binary Group 5 in the 'Stream()' method and reading back 'data.linearAccelBody' in the 'PublishData' method. However on visualising the acceleration in Rviz and also on looking at the contents of the topics, it seemed that the values are quite noisy even when the sensor is standing still. 

I have set the VPE to be enabled and imu_compensated to be true. Even the non-gravity compensated acceleration seemed to be noisy on rviz. I have not been able to use the acceleration measurements from this IMU in my sensor fusion algorithm because of such noisy data. 

Am I missing out on enabling certain flags/settings?

kartikmohta, 2017-10-31:
Can you plot the acceleration values and show the amount of noise?

shubham294, 2017-11-01:
On keeping the sensor still, the peak-to-peak jitter in acceleration value is within 0.03-0.08 m/s^2. Our application has a theoretical maximum of around 0.50 m/s^2.  Given the small value of peak acceleration, the idle jitter is quite higher in comparison.

kartikmohta, 2017-11-01:
Here is a plot of the data I get when the IMU is stationary:
[image]

Seems like the noise is similar to what you described. I believe that this amount of noise is pretty standard.

closed 2017-11-07 by shubham294