KumarRobotics/imu_vn_100 #18: IMU rate consistently lower than setting

arazdan, 2017-11-15:
Hello, I have been having an issue where the IMU rate is consistently lower than what I set it at. For example, when I set the IMU to run at a rate of 80 Hz, rostopic hz /imu/imu shows that messages are only being published at a rate of approximately 56 Hz. Similarly, when I try to set the frequency to 200 Hz, messages are published at a rate of 144 Hz. I am setting the serial baudrate to 921600, which I believe should be plenty for receiving messages at any of the rates I've been experimenting with, and it is possible to increase the IMU rate by setting it to higher numbers, but the output rates are not consistent with the ones that have been set. 

I've tried the FTDI fix in the FAQ section, but it doesn't seem to solve the problem, nor does setting sync_rate to 0. Has anyone else encountered a similar problem and come up with a solution? Thanks in advance for the help.

snakehaihai, 2017-11-21:
type below in a terminal

# Reduce latency in the FTDI serial-USB kernel driver to 1ms
# This is required due to https://github.com/torvalds/linux/commit/c6dce262
for file in $(ls /sys/bus/usb-serial/devices/); do
  value=`cat /sys/bus/usb-serial/devices/$file/latency_timer`
  if [ $value -gt 1 ]; then
    echo "Setting low_latency mode for $file"
    sudo sh -c "echo 1 > /sys/bus/usb-serial/devices/$file/latency_timer"
  fi
done

ke-sun, 2017-11-30:
As @snakehaihai pointed out, the solution is provided in the FAQ of readme in this repo:
https://github.com/KumarRobotics/imu_vn_100#faq

closed 2017-11-30 by ke-sun