KumarRobotics/imu_vn_100 #20: Opertation time out.

Garfield-kh, 2018-07-12:
[image]
After make the ROS work space, the first try of IMU VN100T is workable with my own DIY USB-IMU cable.
Then I use the official USB cable and plug in to windows OS software to test its performance, the test is ok.
However, after I re-try with the ROS launch. the /imu/imu has no new message anymore.
I try to change the Baud Rate to meet with the Window OS one. but still failed.
Does Anyone met the same problem as me?

kartikmohta, 2018-07-13:
Do you have multiple USB-serial devices connected? If yes, can you make sure that the IMU is connected to the one corresponding to `/dev/ttyUSB0`?

Garfield-kh, 2018-07-13:
I will type "sudo chmod a+rw /dev/ttyUSB0" to change the imu permission everytime. so the ttyUSB0 should be the right one. 
I am checking if it's my DIY cable problem. but it was workable before I try the IMU in windows official software.

Garfield-kh, 2018-07-13:
one wire inside the cable damaged. <.< I try repairing it and see. XD.
Thank you.

kartikmohta, 2018-08-16:
Closing this as it seems that you found the issue and fixed it.

closed 2018-08-16 by kartikmohta