KumarRobotics/imu_vn_100 #23: VN: Permission Denied

Elow13, 2018-08-15:
I'm getting a permission denied error when I'm running the launch file. I've changed the baud rate to 921600 and the serial to 1 in line 17, but it's still giving me the error. I'm using an FTDI cable to connect (not just the standard IMU cable), but it looks like it's outputting correct data since it's not getting operation timeouts. Any ideas?

kartikmohta, 2018-08-15:
The node should stop if you get a `VN:Permission denied` message, so I don't think you would be getting any output data after that error. Can you copy paste the full log of the launch?

nerdyneptunian, 2018-08-15:
I don't think I'm getting any data to the node after the error either since the node is closed, it's more of I believe that the computer is getting IMU serial data at least. Here is the log launch:

``roslaunch imu_vn_100 vn_100_cont.launch 
... logging to /home/ebl13/.ros/log/a493d828-a0b9-11e8-8b9b-a402b99bddf4/roslaunch-ebl13-Riptide-14743.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ebl13-Riptide:36921/

SUMMARY
========

PARAMETERS
 * /imu/baudrate: 921600
 * /imu/binary_async_mode: 1
 * /imu/binary_output: True
 * /imu/enable_mag: True
 * /imu/enable_pres: True
 * /imu/enable_rpy: False
 * /imu/enable_temp: True
 * /imu/frame_id: imu
 * /imu/imu_compensated: False
 * /imu/imu_rate: 100
 * /imu/port: /dev/ttyUSB0
 * /imu/sync_pulse_width_us: 1000
 * /imu/sync_rate: 20
 * /imu/vpe/enable: True
 * /imu/vpe/filtering_mode: 1
 * /imu/vpe/heading_mode: 1
 * /imu/vpe/tuning_mode: 1
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    imu (imu_vn_100/imu_vn_100_cont_node)

auto-starting new master
process[master]: started with pid [14753]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a493d828-a0b9-11e8-8b9b-a402b99bddf4
process[rosout-1]: started with pid [14766]
started core service [/rosout]
process[imu-2]: started with pid [14783]
[ INFO] [1534357983.007971110]: Sync out rate: 20
[ INFO] [1534357983.508779636]: Connected to device at /dev/ttyUSB0
[ INFO] [1534357983.528329522]: Default serial baudrate: 115200
[ INFO] [1534357983.528381891]: Set serial baudrate to 921600
[ INFO] [1534357984.564893763]: Connected to device at /dev/ttyUSB0
[ INFO] [1534357984.595156070]: New serial baudrate: 921600
[ INFO] [1534357984.610562737]: Fetching device info.
[ INFO] [1534357984.630900692]: Model number: VN-100T-CR
[ INFO] [1534357984.646174142]: Hardware revision: 7
[ INFO] [1534357984.661466477]: Serial number: 0100037257
[ INFO] [1534357984.676709077]: Firmware version: 2.1.0.0
[ INFO] [1534357984.676753069]: Set Synchronization Control Register (id:32).
[ INFO] [1534357984.707112040]: Default VPE enable: 1
[ INFO] [1534357984.707160128]: Default VPE heading mode: 1
[ INFO] [1534357984.707173070]: Default VPE filtering mode: 1
[ INFO] [1534357984.707182852]: Default VPE tuning mode: 1
[ INFO] [1534357984.707199111]: Setting VPE MagnetometerBasicTuning BaseTuning (4.000000, 4.000000, 4.000000)
[ INFO] [1534357984.707213760]: Setting VPE MagnetometerBasicTuning AdaptiveTuning (5.000000, 5.000000, 5.000000)
[ INFO] [1534357984.707229075]: Setting VPE MagnetometerBasicTuning AdaptiveFiltering (5.500000, 5.500000, 5.500000)
[ INFO] [1534357984.722511696]: Setting VPE AccelerometerBasicTuning BaseTuning (5.000000, 5.000000, 5.000000)
[ INFO] [1534357984.722548056]: Setting VPE AccelerometerBasicTuning AdaptiveTuning (3.000000, 3.000000, 3.000000)
[ INFO] [1534357984.722563899]: Setting VPE AccelerometerBasicTuning AdaptiveFiltering (4.000000, 4.000000, 4.000000)
[ INFO] [1534357984.773274984]: Streaming #1: BG1_QTN|BG1_SYNC_IN_CNT|BG1_IMU
[ INFO] [1534357984.773316025]: Streaming #3: BG3_UNCOMP_MAG|BG3_TEMP|BG3_PRES
[ INFO] [1534357984.796832363]: /imu: VN: Permission denied
[imu-2] process has finished cleanly
log file: /home/ebl13/.ros/log/a493d828-a0b9-11e8-8b9b-a402b99bddf4/imu-2*.log

nerdyneptunian, 2018-08-15:
I don't think I'm getting any data to the node after the error either since the node is closed, it's more of I believe that the computer is getting IMU serial data at least. Here is the log launch:

SUMMARY

PARAMETERS

    /imu/baudrate: 921600
    /imu/binary_async_mode: 1
    /imu/binary_output: True
    /imu/enable_mag: True
    /imu/enable_pres: True
    /imu/enable_rpy: False
    /imu/enable_temp: True
    /imu/frame_id: imu
    /imu/imu_compensated: False
    /imu/imu_rate: 100
    /imu/port: /dev/ttyUSB0
    /imu/sync_pulse_width_us: 1000
    /imu/sync_rate: 20
    /imu/vpe/enable: True
    /imu/vpe/filtering_mode: 1
    /imu/vpe/heading_mode: 1
    /imu/vpe/tuning_mode: 1
    /rosdistro: kinetic
    /rosversion: 1.12.13

NODES
/
imu (imu_vn_100/imu_vn_100_cont_node)

auto-starting new master
process[master]: started with pid [14753]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a493d828-a0b9-11e8-8b9b-a402b99bddf4
process[rosout-1]: started with pid [14766]
started core service [/rosout]
process[imu-2]: started with pid [14783]
[ INFO] [1534357983.007971110]: Sync out rate: 20
[ INFO] [1534357983.508779636]: Connected to device at /dev/ttyUSB0
[ INFO] [1534357983.528329522]: Default serial baudrate: 115200
[ INFO] [1534357983.528381891]: Set serial baudrate to 921600
[ INFO] [1534357984.564893763]: Connected to device at /dev/ttyUSB0
[ INFO] [1534357984.595156070]: New serial baudrate: 921600
[ INFO] [1534357984.610562737]: Fetching device info.
[ INFO] [1534357984.630900692]: Model number: VN-100T-CR
[ INFO] [1534357984.646174142]: Hardware revision: 7
[ INFO] [1534357984.661466477]: Serial number: 0100037257
[ INFO] [1534357984.676709077]: Firmware version: 2.1.0.0
[ INFO] [1534357984.676753069]: Set Synchronization Control Register (id:32).
[ INFO] [1534357984.707112040]: Default VPE enable: 1
[ INFO] [1534357984.707160128]: Default VPE heading mode: 1
[ INFO] [1534357984.707173070]: Default VPE filtering mode: 1
[ INFO] [1534357984.707182852]: Default VPE tuning mode: 1
[ INFO] [1534357984.707199111]: Setting VPE MagnetometerBasicTuning BaseTuning (4.000000, 4.000000, 4.000000)
[ INFO] [1534357984.707213760]: Setting VPE MagnetometerBasicTuning AdaptiveTuning (5.000000, 5.000000, 5.000000)
[ INFO] [1534357984.707229075]: Setting VPE MagnetometerBasicTuning AdaptiveFiltering (5.500000, 5.500000, 5.500000)
[ INFO] [1534357984.722511696]: Setting VPE AccelerometerBasicTuning BaseTuning (5.000000, 5.000000, 5.000000)
[ INFO] [1534357984.722548056]: Setting VPE AccelerometerBasicTuning AdaptiveTuning (3.000000, 3.000000, 3.000000)
[ INFO] [1534357984.722563899]: Setting VPE AccelerometerBasicTuning AdaptiveFiltering (4.000000, 4.000000, 4.000000)
[ INFO] [1534357984.773274984]: Streaming #1: BG1_QTN|BG1_SYNC_IN_CNT|BG1_IMU
[ INFO] [1534357984.773316025]: Streaming #3: BG3_UNCOMP_MAG|BG3_TEMP|BG3_PRES
[ INFO] [1534357984.796832363]: /imu: VN: Permission denied
[imu-2] process has finished cleanly
log file: /home/ebl13/.ros/log/a493d828-a0b9-11e8-8b9b-a402b99bddf4/imu-2*.log

kartikmohta, 2018-08-16:
Just to make sure, are you really connected to serial port 1 (pins 2 and 3) since the that port requires special cables to do the voltage level conversion from RS-232 levels to something that the USB can handle. Most likely you've connected the FTDI cable to pins 8 and 9 on the device, which are serial port 2. Do you get the error even if you set the `binary_async_mode` to 2?

nerdyneptunian, 2018-08-16:
Oh I had switched to the TTL and not changed the binary sync to 2!  It works now! Thank you!

closed 2018-08-16 by nerdyneptunian

krish007k, 2019-04-30:
hello sir!
I did the changes all u mentioned. but I am still getting the same error "permission denied".