KumarRobotics/imu_vn_100 #44: VN permission denied error when enabling binary mode

berndpfrommer, 2021-07-10:
Using the default launch file I get a permission denied error. The driver permissions look fine, and it sets the baud rate as expected. This call triggers the Permission denied error:

      VnEnsure(vn100_setBinaryOutput1Configuration(
          &imu_, BINARY_ASYNC_MODE_SERIAL_2, kBaseImuRate / imu_rate_,
          BG1_QTN | BG1_IMU | BG1_MAG_PRES | BG1_SYNC_IN_CNT | BG1_TIME_SYNC_IN,
          BG3_NONE, BG5_NONE, true));

I can operate the device in ASCII mode and get data just fine. Any idea what the problem could be?

Here is the ROS driver output:
```
[ INFO] [1625943184.351130705]: Sync out rate: 20
[ INFO] [1625943184.862031633]: Connected to device at /dev/ttyUSB0
[ INFO] [1625943184.885040377]: Default serial baudrate: 115200
[ INFO] [1625943184.885198288]: Set serial baudrate to 921600
[ INFO] [1625943185.919464869]: Connected to device at /dev/ttyUSB0
[ INFO] [1625943185.949322535]: New serial baudrate: 921600
[ INFO] [1625943185.966449729]: Fetching device info.
[ INFO] [1625943185.981300794]: Model number: VN-100T-CR
[ INFO] [1625943185.997282869]: Hardware revision: 7
[ INFO] [1625943186.013351897]: Serial number: 0100056785
[ INFO] [1625943186.030187364]: Firmware version: 2.1.0.0
[ INFO] [1625943186.030270266]: Set Synchronization Control Register (id:32).
[ INFO] [1625943186.101733960]: /imu: VN: Permission denied

```

fcladera, 2021-07-14:
Hi @berndpfrommer, were you able to figure out this issue?

berndpfrommer, 2021-07-14:
Nope. Got access to VectorNav's online documents but haven't looked at them yet. I'm wondering if it has to do with the firmware being too old or too recent.

fcladera, 2021-07-14:
I am facing the same problem with our current VN-100T-CR. I'll take a look into the firmware, thanks for the hint!